foxglove/studio #6706: Images broken on studio 1.68, PopOS 22.04 (Linux x86_64), Nvidia driver  535.104.05

venabled, 2023-08-29:
**Description**

Upgraded from v1.65 to v1.68, and Image panel no longer shows images working. Rostopic info on the image topics shows foxglove studio or rosbridge subscribed to those topics, but image panel is blank. No camera info topic is selected. Raw image data is available in the `Raw` panels. Downgrading to 1.65 fixes it.

- Version: 1.68
- Platform: Linux (AMD64) Desktop, PopOS 22.04 LTS, NVIDIA gfx version  535.104.05
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS1 native, ROS1 rosbridge

**Steps To Reproduce**
```
sudo apt update && sudo apt upgrade
```

**Expected Behavior**
Expect images to show up in image panel.

jtbandes, 2023-08-29:
This sounds like it is probably related to https://github.com/foxglove/studio/pull/6665. Please double check the `step` value in your raw images and ensure it is correct. You should see an error message in the panel settings with more details.

venabled, 2023-08-30:
It also doesn't work with compressedImage which is just jpeg, which is
interesting.

Will double check here with the step when I get a moment.

jtbandes, 2023-08-30:
Could you share if there are any warnings or errors that appear in the dev console? (ctrl+shift+i)

venabled, 2023-08-30:
For reference these are mono8,mono16 images for the most part. Step == 640 for mono8, 1280 for mono16.

venabled, 2023-08-30:
[image]

jtbandes, 2023-08-30:
Do you get the same error on studio.foxglove.dev in Chrome?

venabled, 2023-08-30:
No, I'm able to see both 16 and 8 bit mono as well as compressed using the
web instance.

jtbandes, 2023-08-30:
What version of chrome is that?

venabled, 2023-08-30:
`Version 116.0.5845.140 (Official Build) (64-bit)`

jtbandes, 2023-09-12:
Could you try with Chrome 114? (That's the same version used by the desktop app currently.) Might also be worth trying 117 (latest beta) and 118 (latest dev, will be used in next electron release). See also: https://chromestatus.com/roadmap, https://www.electronjs.org/docs/latest/tutorial/electron-timelines

venabled, 2023-09-20:
Still broken on 1.71 

[image]

jtbandes, 2023-09-20:
Yeah, we haven't changed the version of Electron/Chromium used in the app recently. This seems like an incompatibility with your drivers or hardware. I would be interested to learn if older or newer versions of the Chrome browser have different behavior – we can upgrade Electron again, but we previously had reports of other Linux users running into complete app crashes on the newer version (https://github.com/foxglove/studio/issues/6654) so I am hesitant to upgrade unless we can get confirmation that things are working.

venabled, 2023-09-20:
Remind me again why I have an NVIDIA card on Linux.

emersonknapp, 2023-09-26:
Just adding another datapoint that we're also seeing this issue

Version: 1.71.0
Platform:
```
$ uname -a
Linux my-hostname 6.2.0-33-generic #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 10:33:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
```

```
NVIDIA-SMI 535.104.05             Driver Version: 535.104.05   CUDA Version: 12.2
NVIDIA GeForce RTX 4070
```

Data source: websocket

Downgrading to 1.65 solves it for now.

jtbandes, 2023-10-02:
@emersonknapp Thanks for adding more details – could you also confirm whether this occurs with the sample dataset (and with which panels/layouts) or share an example data file that reproduces the issue?

venabled, 2023-10-11:
This is fixed with 1.72.0, NVIDIA drivers `535.113.01`

closed 2023-10-11 by venabled

acowley, 2023-11-04:
I believe I am also seeing this problem with Foxglove Studio 1.75.0 on Ubuntu 20.04.6 and Nvidia drivers 535.129.03. Symptoms are blank Image and 3D panels.

```
THREE.WebGLProgram: Shader Error 0 - VALIDATE_STATUS false

Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Oct 19 2023. Please check build information of source that generated the binary.
```

jtbandes, 2023-11-09:
I think this will be fixed in https://releases.electronjs.org/release/v27.0.4