foxglove/studio #6713: When I use convertTo field for a subscribed message, allMessages is empty but currentMessage is present

emresiimseek, 2023-08-31:
When I use convertTo field for a subscribed message, allMessages is empty but currentMessage is present.
The relevant converter has been registered.
allMessages appear when convertTo field is deleted.

- Version:1.68.0
- Platform:Web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**
[image]

```javascript
  useLayoutEffect(() => {
    context.onRender = (render_state: RenderState, done) => {
      setRenderDone(() => done);

      if (render_state.currentFrame) {
        setLocationFix(render_state.currentFrame?.[0]?.message);
      }
    };

    context.watch('currentFrame');
    context.watch('allFrames');
    context.subscribe([{ topic: '/message/message', convertTo: 'foxglove.LocationFix' }]);
  }, [context]);
``` 

**Expected Behavior**

foxymiles, 2023-09-01:
What happens if you set `preload: true` in your `subscribe` call?

foxymiles, 2023-09-12:
Can you please try setting `preload: true` in your subscribe call? I think you will see `allMessages` populated if you do this.

foxymiles, 2023-09-18:
I believe this is working as intended and you'll get the result you want if you pass the `preload: true` flag to your subscribe call. I'll close this for now and we can reopen it if that's not the case.

closed 2023-09-18 by foxymiles