foxglove/studio #6728: Disable SSL checking enforcement on .dev TLDs or provide guidance :)

iliabaranov, 2023-09-03:
**Description**

Hello!

We run our robots with .dev TLDs, for example: robot1.polymathrobotics.dev

However, Foxglove seems to not be able to stream data in ROS1, ROS2, or websockets mode when using this as a hostname.
Some details here in slack:
https://foxglove.slack.com/archives/C028UEY858S/p1693776156190629

TLDR: I get "net::ERR_SSL_PROTOCOL_ERROR" when connecting to machines with the above .dev TLD

**Expected Behavior**

I'd suggest, since Foxglove is not intended to be a consumer browser or similar, to either disable SSL enforcement, or to clarify steps on where/how the system is looking for SSL certs.

amacneil, 2023-09-05:
The whole `.dev` top level domain (TLD) has [HSTS enabled](https://endtimes.dev/dev-and-hsts-preload/), which enforces HTTPS. This means that it is blanket not possible to host HTTP content on `.dev` domains. If you still want to use HTTP I would recommend using a different TLD.

I'm not sure whether it is possible to disable HSTS enforcement, but even if it were possible, it feels a bit icky intentionally reducing security. Foxglove isn't a consumer browser, but we do rely heavily on web technologies, and the web standards effectively specify that you cannot use HTTP with `.dev`.

closed 2023-09-05 by amacneil

jtbandes, 2023-09-05:
Just to be clear, the app does support `wss://` connections, if you set up your websocket server with SSL support.

iliabaranov, 2023-09-05:
To follow up: With a proper signed certificate, native Foxglove studio app does work, hooray :)

However, it is REALLY slow, approx. ~4x slower update rate than just normal RVIZ.
Is this expected? IIRC it was not this slow before wss and TLS were used?

amacneil, 2023-09-06:
I wouldn't expect any noticeable performance impact from TLS. 

If you connect to the same websocket endpoint using vanilla http (i.e. `ws://`) and the IP address directly (instead of your `.dev` domain), what's it like?