foxglove/studio #6736: File open options do nothing

MattOslin, 2023-09-05:
**Description**

File > `Open...`
File > `Open local file...`
File > `Open connection...`

No longer work. Only able to get the open dialog by opening a new window.

- Version: 1.68.0
- Platform: macos 13.3.1 (a)
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

Open studio, load a local file or connection. Click File > Open. Nothing occurs.

**Expected Behavior**

A dialog box opens to allow the user to open a file or connection.

jtbandes, 2023-09-05:
What browser are you using? 

> Only able to get the open dialog by opening a new window.

Does this mean the issue resolves itself once you refresh or restart your browser?

MattOslin, 2023-09-06:
> What browser are you using?
> 
> > Only able to get the open dialog by opening a new window.
> 
> Does this mean the issue resolves itself once you refresh or restart your browser?

There is no browser involved, I am using the macos app. The issue does not resolve itself, I can no longer open logs without opening new windows.

MattOslin, 2023-09-06:
I have found a workaround, this menu works for me, but only when not in fullscreen mode. In fullscreen mode clicking it does nothing (it used to work).
[image]

MattOslin, 2023-09-06:
In fullscreen mode there is no hover-over response or click response from the foxglove logo
[image]

jtbandes, 2023-09-06:
I am able to reproduce the native File menu not working. However, I am not able to reproduce the issue with the foxglove logo not being clickable in fullscreen mode. Which macOS version are you using?

MattOslin, 2023-09-07:
> I am able to reproduce the native File menu not working. However, I am not able to reproduce the issue with the foxglove logo not being clickable in fullscreen mode. Which macOS version are you using?

13.3.1 (a)

PR #6773 (2023-09-08): Fix File > Open menu items not working
**User-Facing Changes**
Fixed an issue where some File menu options didn't work on macOS.

**Description**
Related to changes made in https://github.com/foxglove/studio/pull/6273, https://github.com/foxglove/studio/pull/6698 & https://github.com/foxglove/studio/pull/6699
Fixes #6736 

Clicking "Open", "Open local file", or "Open connection" in the system menu didn't work, instead logging this error:

```
(node:31329) UnhandledPromiseRejectionWarning: TypeError: Cannot read properties of undefined (reading 'webContents')
    at simulateUserClick (webpack-i [... cut]
commit 7eb2f08d (2023-09-08): Fix File > Open menu items not working (#6773)

**User-Facing Changes**
Fixed an issue where some File menu options didn't work on macOS.

**Description**
Related to changes made in https://github.com/foxglove/studio/pull/6273,
https://github.com/foxglove/studio/pull/6698 &
https://github.com/foxglove/studio/pull/6699
Fixes #6736 

Clicking "Open", "Open local file", or "Open connection" in the system
menu didn't work, instead logging this error:

```
(node:31329) UnhandledPromiseRejectionWarning: TypeError: Cannot read properties of undefined (reading 'webContents')
    at simulateUserClick ( [... cut]
closed 2023-09-08 by jtbandes