foxglove/studio #6821: LinePrimitive not compatible between protobuf messages and User Scripts

aharmat, 2023-09-13:
I'm trying to modify a SceneUpdate message with a User Script and then return a new SceneUpdate message, but I encountered the following error:
```
Type '{ deletions: { timestamp: { sec: number; nsec: number; }; type: number; id: string; }[]; entities: { timestamp: { sec: number; nsec: number; }; frame_id: string; id: string; lifetime: { sec: number; nsec: number; }; ... 9 more ...; models: { ...; }[]; }[]; }' is not assignable to type 'SceneUpdate'.
  Types of property 'entities' are incompatible.
    Type '{ timestamp: { sec: number; nsec: number; }; frame_id: string; id: string; lifetime: { sec: number; nsec: number; }; frame_locked: boolean; metadata: { key: string; value: string; }[]; arrows: { pose: { position: { ...; }; orientation: { ...; }; }; ... 4 more ...; color: { ...; }; }[]; ... 6 more ...; models: { ...;...' is not assignable to type 'SceneEntity[]'.
      Type '{ timestamp: { sec: number; nsec: number; }; frame_id: string; id: string; lifetime: { sec: number; nsec: number; }; frame_locked: boolean; metadata: { key: string; value: string; }[]; arrows: { pose: { position: { ...; }; orientation: { ...; }; }; ... 4 more ...; color: { ...; }; }[]; ... 6 more ...; models: { ...;...' is not assignable to type 'SceneEntity'.
        Types of property 'lines' are incompatible.
          Type '{ type: number; pose: { position: { x: number; y: number; z: number; }; orientation: { x: number; y: number; z: number; w: number; }; }; thickness: number; scale_invariant: boolean; points: { x: number; y: number; z: number; }[]; color: { ...; }; colors: { ...; }[]; indices: Uint32Array; }[]' is not assignable to type 'LinePrimitive[]'.
            Type '{ type: number; pose: { position: { x: number; y: number; z: number; }; orientation: { x: number; y: number; z: number; w: number; }; }; thickness: number; scale_invariant: boolean; points: { x: number; y: number; z: number; }[]; color: { ...; }; colors: { ...; }[]; indices: Uint32Array; }' is not assignable to type 'LinePrimitive'.
              Types of property 'indices' are incompatible.
                Type 'Uint32Array' is missing the following properties from type 'number[]': pop, push, concat, shift, and 5 more.(2322)
```

I believe the problem is similar to https://github.com/foxglove/studio/pull/6785.

- Version: 1.70
- Platform: MacOS

**Steps To Reproduce**
Create a User Script as follows and load a data source where `/viz/input` contains protobuf messages of type `SceneUpdate`. I have only tested this with LinePrimitive entities but other primitives might trigger similar problems.

```
import { Input } from "./types";
import { SceneUpdate } from "@foxglove/schemas";

export const inputs = ["/viz/input"];
export const output = "/viz/output";

export default function script(event: Input<"/viz/input">): SceneUpdate  {
  return event.message;
}
```

**Expected Behavior**
The input SceneUpdate type is compatible with the one that the User Script loads from @foxglove/schemas.

jtbandes, 2023-09-25:
Could you give this version a try and see if it works for you?

https://github.com/foxglove/studio/pull/6890
https://studio-cnrqk4c1s.foxglove.party/

aharmat, 2023-09-25:
It works for me, thank you.

PR #6890 (2023-09-28): Add support for TypedArrays in User Scripts using @foxglove/schemas
**User-Facing Changes**
Fixed a type incompatibility issue with User Scripts that import from @foxglove/schemas.

**Description**
Fixes #6821 
Fixes FG-4976
Depends on https://github.com/foxglove/schemas/pull/128
Also updated JSON message datatype extraction to treat integer arrays as `Uint32Array` rather than `Float64Array`; this fixed a similar issue as #6821 with an example json-based mcap I had lying around.
commit 7ad78eaa (2023-09-28): Add support for TypedArrays in User Scripts using @foxglove/schemas (#6890)

**User-Facing Changes**
Fixed a type incompatibility issue with User Scripts that import from
@foxglove/schemas.

**Description**
Fixes #6821 
Fixes FG-4976
Depends on https://github.com/foxglove/schemas/pull/128
Also updated JSON message datatype extraction to treat integer arrays as
`Uint32Array` rather than `Float64Array`; this fixed a similar issue as
#6821 with an example json-based mcap I had lying around.
closed 2023-09-28 by jtbandes