foxglove/studio #6823: User scripts not being run in 1.70 release

aharmat, 2023-09-14:
User scripts are not being executed in the 1.70 release of Studio. Output topic names don't appear in the relevant panels, and `log` statements put into the code don't print anything.

- Version: 1.70
- Platform: 1.68
- Data source: mcap, foxglove data storage

**Steps To Reproduce**
Try to run any user script in 1.70.

**Expected Behavior**
User scripts execute as with previous releases.

jtbandes, 2023-09-14:
I'm not able to reproduce this issue. Could you share more detailed steps to reproduce, such as sample data and a layout?

[image]

aharmat, 2023-09-15:
I have experimented with the following layouts:
- Creating a new one
  - While signed in to my org
  - While signed out
- Modifying one from my organization
  - Saving a local copy
  - Not saving a local copy

An in all cases, what I have reduced it to is that the inclusion of any type from `@foxglove/schemas` in any user script stops all user scripts from running. For example, adding `import { LocationFix } from "@foxglove/schemas";` to the hello world script stops it from running. Or for another example, create two scripts, one the hello world sample, and another that only contains `import { Color } from "@foxglove/schemas";`. The hello world sample runs if you comment out the one import line in the 2nd script, but not if you have it uncommented.

jtbandes, 2023-09-15:
Thanks, that is helpful, I'm able to reproduce the issue now.

jtbandes, 2023-09-22:
Fixed in https://github.com/foxglove/studio/pull/6862

closed 2023-09-22 by jtbandes