foxglove/studio #6824: Missing code

qinian9584, 2023-09-14:
**Description**

Excuse me, These functions，Where is the source code located ？

main branch ，Not found

[image]

- Version:
- Platform:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

**Expected Behavior**

jtbandes, 2023-09-14:
Duplicate of https://github.com/orgs/foxglove/discussions/827

closed 2023-09-14 by jtbandes

amacneil, 2023-10-02:
The multi-layouts feature is undergoing significant changes in our closed source edition, and now relies on our cloud service. Previously, there was a lot of complexity around managing both local and cloud layouts, and issues when signing in/out of accounts. We decided to remove all cloud logic from the open source codebase, so we are no longer maintaining two versions.

You can still access the multi-layouts feature for free in our [official binaries](https://foxglove.dev/download). If you wish to stick to open source, there is also still an [active PR](https://github.com/foxglove/studio/pull/5987) to support loading a layout from URL.