foxglove/studio #6831: Image doesn'

jerry-xie-000, 2023-09-15:
**Description**

- Version:
- Platform:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

**Expected Behavior**

jtbandes, 2023-09-15:
I'm not able to download the file from baidu. Can you upload it to GitHub by attaching it to a comment here? You will need to create a .zip or .tar.gz file first. https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files

jerry-xie-000, 2023-09-18:
The mcap file:
[20230915-150145_0.zip](https://github.com/foxglove/studio/files/12644247/20230915-150145_0.zip)

snosenzo, 2023-09-20:
I'm not able to reproduce with your sample file. Everything seems to be working properly on main locally, I'm not seeing the error you're getting on the image topic.
[image]
using this layout
[foxglove-layout.json.zip](https://github.com/foxglove/studio/files/12676277/foxglove-layout.json.zip)
Would you mind sharing the layout you used?
Also I see you have `sync annotations` and an annotation topic toggled on in your settings in the screenshot, but it doesn't look like there are any `ImageAnnotations` on that topic in the file you shared. When `Sync annotations` is enabled it will prevent the rendering of the image except when the panel has received a message from each Annotation topic with the exact same timestamp as the image. This could explain why some of the image panels aren't showing images, but it wouldn't explain the error you're getting on the "Image topic" about the "source width" (not sure what would've caused that).

jerry-xie-000, 2023-09-21:
Ok, thanks. Now I understand the "Sync annotations" feature.

I use your layout, and it works.  

Here is my layout, I think it can reproduce the "source width" issue.
[space920.zip](https://github.com/foxglove/studio/files/12679516/space920.zip)

snosenzo, 2023-09-21:
@jerry-xie-000 I'm still not seeing the "source width" issue with your layout. The image panel showing the `/od/front` message has sync annotations turned on still, which is why it's not showing. That error could be because of errant data within a certain source. Are you still encountering this issue?

jerry-xie-000, 2023-09-22:
I reproduce this mcap file on three other computers:  both Win10 and Win11, however the issue is still. 

I also redownload the new version 1.71, the issue is still too.

It seems the issue is easy to be reproduced. Maybe you can try on other computer?

snosenzo, 2023-09-22:
@jerry-xie-000  thanks for the info. I'll test on some other systems. Just a few more questions: are you encountering this error in the desktop version of the app or in the web version or both? If you're using the web version, are you also using chrome?

snosenzo, 2023-09-22:
Ah I just tested on our desktop app and I do see the error on both Win10 and Mac. It doesn't look like it shows up in the browser app though. Will investigate and get it fixed. Thanks for the additional info.

snosenzo, 2023-09-26:
Looks like it was actually fixed in this [PR](https://github.com/foxglove/studio/pull/6885). Which is in production on the web version at studio.foxglove.dev, and will be in the next desktop release (after 1.71.0). Feel free to open this back up if you're still seeing it though

closed 2023-09-26 by snosenzo