foxglove/studio #6832: Live plot not showing anything

BPelourdeau, 2023-09-15:
**Description**
While in live, the plots don't work anymore

- Version: 1.69 and 1.70
- Platform: Windows 10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): live

**Steps To Reproduce**

Just connect to a robot and plot something. I've seen the same issue on another computer. The problem is not there for version 1.67. I haven't tested version 1.68, so I can't tell for this one.

**Expected Behavior**
Same as before, to see what we plot :)
[image]
[image]

BPelourdeau, 2023-09-15:
First image is 1.67 and second is 1.70

cfoust, 2023-09-15:
Hey! Thanks for reporting. I'm looking into this 🤔

r0gi, 2023-09-17:
I've had the same issue. the plot panels have much better performance now, but live plots don't show anything. Ubuntu 20.04 and 22.04, Foxglove Studio v1.70. Plots from bag files work fine.

BPelourdeau, 2023-09-19:
I totally forgot to mention I was connecting to a ROS1 source... Good job on that catch you are awesome !

commit b2e67a77 (2023-09-18): Fix for plots on live ROS1 sources (#6845)

**User-Facing Changes**
None.

**Description**
The messages produced by `LazyMessageReader` are empty after being
`postMessage`d; we must call `toObject` on them.

Fixes #6832