foxglove/studio #6842: Black screen while decode a none standard ROS data from rosbag

qdhuxp, 2023-09-18:
**Description**
I have a rosbag with a lidar point cloud topic which is encoded in none standard  ROS pointcloud definition. I want to decode it through Userscript panel, but get black screen while playing data. And from chrome dev tool, I can see an error with "Paused before potential out-of-memory crash".

- Version: 1.69.0
- Platform: Windows 10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbag file

**Steps To Reproduce**

1. Open Studio, export layout below:
  [Sample_ Nuscenes_blackscreen.zip](https://github.com/foxglove/studio/files/12645104/Sample_.Nuscenes_blackscreen.zip)
  This layout contains a UserScript panel to sub lidar topic and output a "hello: world" string.
2. Play the rosbag below for a while, Studio will get stuck and be black screen.
    [file is large, uploading....]

**Expected Behavior**

qdhuxp, 2023-09-19:
I don't know how to share issue rosbag(about 590M), tried many ways but blocked by network, any ideas?

jtbandes, 2023-09-19:
Can you trim the bag start/end time so it has less data? Or does the issue only occur with a large bag?
You could use google drive to upload it.

qdhuxp, 2023-09-19:
Let me upload the issue video first. The rosbag only contains 2 ropics, one is lidar topic which data size is about 30M/frame. 
https://www.aliyundrive.com/s/8EWu6EGJmDG

jtbandes, 2023-09-19:
I can't access these files without creating an account. Is there another service where you can upload them that does not require an account?

qdhuxp, 2023-09-25:
Tried，I could not access other site....

jtbandes, 2023-09-25:
Maybe you can trim the bag down to a smaller size? Does it still show the issue in that case?

defunctzombie, 2023-10-15:
Closing this for lack of response. We need a publicly available file to debug such issues. We are happy to re-visit once you are able to provide an accessible bag file and instructions to reproduce.

closed 2023-10-15 by defunctzombie