foxglove/studio #6844: SceneUpdate Converter does not show anything from the orignal topic (from a bag) but it shows the content if the messages are forwarded from another node

manuelyape, 2023-09-18:
I wrote a converter extension to visualize a custom buonding box message to a SceneUpdate message, to test it I used a bag in which the custom message in published. The problem I'm currently facing is that if I play the bag I'm not able to see anything, the topic appears among the ones can be visualized in the 3D panel, but the boxes are not shown, nevertheless if i run a node that simply forwards the message in another topic, the custom message is correctly converted and visualized in the 3D panel.

- Version: Foxglove Studio 1.70.0
- Platform: Ubuntu 20.04
- Data source: bag file & ROS 1 native

Steps To Reproduce:

The custom message type is called "perception_msgs/Box3DArray" and has the following schema:

**Box3DArray**
```
std_msgs/Header header
Box3D[] obstacles
```
**Box3D**
```
geometry_msgs/Point[4] base_vertices 
geometry_msgs/Pose pose  
geometry_msgs/Point closest_point 
float32 length      
float32 width        
float32 height                          
float32 distance
float32 angle
```

The code of the converter is the following:
```
import { ExtensionContext } from "@foxglove/studio";
import {SceneEntity, CubePrimitive, Point3, Pose} from "@foxglove/schemas";
import { Time } from "@foxglove/schemas/schemas/typescript/Time";

type Header = {
  seq: number,
  stamp: Time,
  frame_id: string
};

type Box3D = {
  header: Header,
  base_vertices: Point3[],
  pose: Pose,
  closest_point: Point3,
  length: number,
  width: number,
  height: number,
  distance: number,
  angle: number
}

type Box3DArray = {
  header: Header,
  obstacles: Box3D[]
}

export function activate(extensionContext: ExtensionContext): void {
  extensionContext.registerMessageConverter({
    fromSchemaName: "perception_msgs/Box3DArray",
    toSchemaName: "foxglove.SceneUpdate",
    converter: (inputMessage: Box3DArray) => {
                  
      var cubes: CubePrimitive[] = [];
      var i = 0;
      for( var box of inputMessage.obstacles) {
        
        var cube: CubePrimitive = {
          pose: {
            position: { 
              x: box.pose.position.x,
              y: box.pose.position.y, 
              z: box.pose.position.z 
            },
            orientation: { 
              x: box.pose.orientation.x,
              y: box.pose.orientation.y,
              z: box.pose.orientation.z,
              w: box.pose.orientation.w 
            },
          },
          size: { 
            x: box.length, 
            y: box.width, 
            z: box.height 
          },
          color: { r: 1, g: 1, b: 1, a: 0.3 },
        }
        cubes.push(cube);
      };
      const sceneEntityMessage: SceneEntity = {
          id: i.toString(),
         timestamp: {
            sec: inputMessage.header.stamp.sec,
            nsec: inputMessage.header.stamp.nsec
          },
          frame_id: inputMessage.header.frame_id,
          lifetime: { sec: 0, nsec: 500000000 },
          frame_locked: false,
          metadata: [],
          arrows: [],
          cubes: cubes,
          spheres: [],
          cylinders: [],
          lines: [],
          triangles: [],
          texts: [],
          models: [],
      };

      return {
        deletions: [],
        entities: [sceneEntityMessage],
      }
    },
  });
}
```

I tested the converter loading the bag as local file in foxglove and also connecting to ROS1 master. Since the message from the bag was not visualized i made some test starting from your Doc examples and finally I tried to run a python node that simply subscribes to the original topic and forwards the custom message in another topic:

```
#!/usr/bin/env python
import rospy
from perception_msgs.msg import Box3DArray 
import std_msgs.msg

pub3 = rospy.Publisher('/my_array_box3d', Box3DArray, queue_size=10)

def callback(data):
    # Box3DArray
    array_box = Box3DArray()
    array_box = data
    # Publish the message
    pub3.publish(array_box)

def listener():
    rospy.init_node('box_forward', anonymous=True)
    sub = rospy.Subscriber('/obstacles_boxes', Box3DArray, callback)
    rospy.spin()

if __name__ == '__main__':
    try:
        listener()
    except rospy.ROSInterruptException:
        pass
```
Expected Behavior:
What I was expecting is that, since the message types and the content are the same, the custom message was correctly converted from both topics, instead only if I run the python node that forwards the message it is visualized in the 3D panel. Both topics appear on the 3D panel as they can be visualized but only the forwarded one (/my_array_box3d) is actually visualized.

jtbandes, 2023-09-18:
Please share the bag file, otherwise I'm not sure how we would be able to reproduce this issue without a lot of custom code to produce messages in the schema you shared.

manuelyape, 2023-09-19:
Hi, of course! Here is the bag: https://mega.nz/file/oDFgiSqB#DdVsPNHSBqD2-N6Z7vB0yw_62D4N11K86FC-uKrbkFs. The topic with the custom message is called "/obstacles_boxes". Thanks!

achim-k, 2023-09-19:
This works without problems for me, I can visualize and see the `/obstacles_boxes` topic when playing back the bag file that you provided.

[Screencast from 19.09.2023 14:36:12.webm](https://github.com/foxglove/studio/assets/9250155/289490bc-b426-46a7-9b2d-f90e56704e21)

Could you share a screencast of the problem? Ideally with the 3D panel settings / raw message panel with the `/obstacle_boxes` topic

manuelyape, 2023-09-19:
I tried to load the bag file in studio but no boxes are shown the same. I tried also to record a bag with the python node (running and stopping it like in the screencast video), I loaded it in studio and again, the same result.
The errors were something like that: The subscription to "/my_array_box3d" will be automatically re-established.

manuelyape, 2023-09-19:
Can you please send me your .foxe extension's compiled code? May my problems be related to my setup (OS, npm, node versions..)?

achim-k, 2023-09-19:
[foxglove.foxglove-example-message-converter-1.1.0.zip](https://github.com/foxglove/studio/files/12664005/foxglove.foxglove-example-message-converter-1.1.0.zip)

manuelyape, 2023-09-19:
Nothing changes, the boxes are not visualized. Any idea on what might be the source of the problem? The OS, ROS? I'll try on another PC to verify if only on the current setup happens.

achim-k, 2023-09-19:
Did you  have any luck with trying it on another PC?

Could you try with a fresh install of Studio? Also, would be best if you make sure that only the message converter in question is loaded e.g. by temporary moving all other extensions in `~/.foxglove-studio` or `~/snap/foxglove-studio` to another directory. It should definitely work then when loading the bag file in Studio.

jtbandes, 2023-09-19:
Any errors in the developer console? (View > advanced)

manuelyape, 2023-09-20:
> Did you have any luck with trying it on another PC?
>
> Could you try with a fresh install of Studio? Also, would be best if you make sure that only the message converter in question is loaded e.g. by [...]

I tried on other PC and it correctly worked, so I followed your suggestions and I uninstalled foxglove-studio, deleted `~/.foxglove-studio` and `~/.config/Foxglove Studio` folders and installing again foxglove and my extensions everything worked!
Thank you for the support!

closed 2023-09-20 by achim-k