foxglove/studio #6893: Zero timestamps are rendered at the beginning of transform history

Woz4tetra, 2023-09-22:
Foxglove renders messages with a timestamp of zero differently from rviz. 

- Version: Tested on 1.60.0 and 1.71.0
- Platform: Ubuntu 20.04.5 LTS focal, Linux 5.15.0-60-generic
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): tested on ROS 1 and rosbridge

**Steps To Reproduce**
My system has a point cloud that publishes every few minutes. I noticed it would publish at the location where the robot used to be. When I relaunched the app or switched layouts the cloud would be rendered in the correct place (the topic is latching). To confirm this behavior, I ran a local test where I publish a simulated cloud after moving the robot some distance away. The catch is I subtracted 10 seconds from the timestamp. Sure enough it renders the cloud where the robot was 10 seconds ago as expected. I ran the same test with a timestamp of zero and it renders the cloud where the robot started.

In this gif, I'm demonstrating this behavior:
[image]
Off screen, every time I press the enter key a new cloud is published in the frame of the robot. In the beginning, the cloud is published in the correct place. However after I drive the robot forward and publish a new cloud, it's rendered where the robot started. I adjust the script to publish with the current time instead of zero and publish a new message. Only then does the cloud render in the correct place.

RViz does not exhibit this behavior. Point clouds with timestamp zero are rendered where the robot currently is in time. This is the expected behavior based on tf2 lookup transform docs. If a timestamp of zero is supplied, it will assume you want the latest transform not the earliest.

**Expected Behavior**
Point clouds with timestamp zero should render at the latest transform not earliest to line up with tf2 lookup transform behavior.

achim-k, 2023-09-26:
> However after I drive the robot forward and publish a new cloud, it's rendered where the robot started. I adjust the script to publish with the current time instead of zero and publish a new message. Only then does the cloud render in the correct place.

You should always publish your messages with a timestamp matching the time of data recording (e.g. at the time the image was taken). While for `tf2`, a timestamp of 0 may mean the latest available message, this only applies for looking up transforms from the tf buffer. It does not imply that you can also publish messages, that you intend to be the most up to date, with a timestamp of 0.

achim-k, 2023-09-27:
@Woz4tetra does the above answer solve your problem or do you still consider this a bug in Studio?

achim-k, 2023-09-27:
I think the ROS behavior was just added for convenience as it avoids waiting for the transform in case it hasn't been received yet. Publishing with 0 timestamp seems odd to me and I don't think that the ROS folks had your use case in mind when deciding to give time 0 a special meaning (for tf lookup). 

In any case, I'm going to close this issue as you could solve your problem by publishing using the current time as timestamp.

closed 2023-09-27 by achim-k