foxglove/studio #6904: Foxglove-studio snap tries to find libraries from previous versions

Guillaumebeuzeboc, 2023-09-25:
**Description**

- Version: 1.71.0
- Platform: Ubuntu 20.04, snap package
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap file

**Steps To Reproduce**
- ```sudo snap install foxglove-studio```
- Launch foxglove-studio and open an MCAP file.
- Wait for the release of an update and the snap to automatically update.
- Launch foxglove-studio and try to open a local file.
- Get the following log:
```
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /snap/foxglove-studio/70/data-dir/icons/Adwaita/16x16/status/image-missing.png: Unable to load image-loading module: /snap/foxglove-studio/46/gnome-platform/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-png.so: /snap/foxglove-studio/46/gnome-platform/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-png.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
[0925/181606.319148:ERROR:scoped_ptrace_attach.cc(27)] ptrace: Operation not permitted (1)
[1]    439149 abort (core dumped)  foxglove-studio
```
(In the case of the log, I am running the revision 70, the revision 46 has been automatically deleted from my computer months ago.)

**Expected Behavior**
Foxglove-studio shouldn't rely on deleted revisions.
This is probably happening because the cache is currently stored in the `SNAP_USER_COMMON` (kept over revisions). Maybe storing it into `SNAP_USER_DATA` (specific to each revision) would fix the issue.

Purging and reinstalling the snap with `sudo snap remove foxglove-studio --purge && sudo snap install foxglove-studio` fixes the issue.

jtbandes, 2023-10-02:
Thanks for the report. I am not very familiar with how the snap data directories are configured. Would you happen to have any guidance on what changes might need to be made to fix the issue? What do you think is being cached that causes it to load a shared library from a previous version? (I'd expect the shared libraries to be part of the installed package, not part of the "caches" per se.) And are there other snaps which ship Electron apps that have already found appropriate workarounds?

(Also, are there any easier steps to reproduce the issue that don't involve waiting for a new version to be released?)

Guillaumebeuzeboc, 2023-10-04:
From what I could see, libraries paths are cached in the file: `~/snap/foxglove-studio/common/.cache/gdk-pixbuf-loaders.cache`.
The problem is that the `~/snap/foxglove-studio/common` is kept over revision. The path `~/snap/foxglove-studio/current` is on the contrary only specific to a revision.

So here I suspect that either the `gdk-pixbuf-loaders.cache` should be placed in a revision specific location, or it should be regenerated at each revision.

To clarify the snap directories:  [snap data and file storage documentation](https://ubuntu.com/robotics/docs/snap-data-and-file-storage).

To reproduce it, multiple revision of the snap must be on the computer (by default 2 are kept over updates). So generating a new snap (by building foxglove-studio) and installing it should generate revision.

The revisions can be listed by:
```
snap list foxglove-studio --all
```
Once multiple revision are available:
- Launch foxglove-studio + open a local mcap file
- `snap revert foxglove-studio`
- Launch foxglove-studio + open a local mcap file -> it should crash.

It could be that this is an electron snap builder limitation.
I am currently limited in time, but I can have a closer look in one month from now.

jtbandes, 2023-10-05:
This might be relevant but I'm not sure: https://forum.snapcraft.io/t/vs-code-quits-crashes/19339

defunctzombie, 2023-10-15:
@Guillaumebeuzeboc Thanks for the additional details - We also are not so intimately familiar with snaps so if you can take a look that would be helpful. We'll also reach out to our contacts at Canonical who helped us ship the Snap and see if they can offer some guidance.

Guillaumebeuzeboc, 2023-10-26:
Hello everyone,

So I dug further today and found out what is the problem.
It's not a foxglove-studio snap specific issue.
The issue would happen to any GUI snap using GDK and reverting.
Since it only affects a rare use case this is not critical.
If it affects someone the current fix would be:
`rm ~/snap/foxglove-studio/current/.last_revision`

I opened [a ticket to report the issue on snapcraft](https://bugs.launchpad.net/snapcraft/+bug/2041182).
Tldr: The cache is only built during the installation and never updated. Meaning that switching to an already installed revision is not going to rebuild its cache.

Since this is not a foxglove specific issue this issue could be closed.

achim-k, 2023-10-26:
Thanks for investigating @Guillaumebeuzeboc!

closed 2023-10-26 by achim-k