foxglove/studio #6914: The legend value of `bigint` type is wrong in Plot Panel

lzcwds, 2023-09-26:
**Description**

in plot panel, the legend value of `bigint` type is wrong
- Version:1.70.0
- Platform: Linux

I try to modify currentValue in `PlotLegendRow.tsx` file, it's only my solution.
```
  const currentValue = useMemo(() => {
    if (!showPlotValuesInLegend) {
      return undefined;
    }
    const timeToCompare = hoverValue?.value ?? currentTime;

    let value;
    for (const pt of correspondingData) {
      if (timeToCompare == undefined || pt.x > timeToCompare) {
        break;
      }
      value = pt.value;
    }
    if (typeof value === "bigint") {
      return value.toString();
    }

    // if (typeof value === "number") {
    //   return parseFloat(value.toFixed(showLabelPrecision));
    // }
    return value;
  }, [
    showPlotValuesInLegend,
    hoverValue?.value,
    currentTime,
    correspondingData,
    // showLabelPrecision,
  ]);
```

**Expected Behavior**
example
[image]

jtbandes, 2023-09-26:
Please provide sample data that we can use to reproduce the issue.

closed 2023-10-03 by jtbandes

jtbandes, 2023-10-03:
Thanks for the report!

PR #6934 (2023-10-03): Use actual point value instead of computed y coordinate in plot legend
**User-Facing Changes**
Fixed an issue where values shown in the Plot legend were sometimes inaccurate.

**Description**
I'm not sure if this ever worked properly...
Fixes #6914
Fixes FG-5129
commit fdabffc6 (2023-10-03): Use actual point value instead of computed y coordinate in plot legend (#6934)

**User-Facing Changes**
Fixed an issue where values shown in the Plot legend were sometimes
inaccurate.

**Description**
I'm not sure if this ever worked properly...
Fixes #6914
Fixes FG-5129