foxglove/studio #6951: Position of line and text are exactly same but those are displayed different position in 3D panel

emresiimseek, 2023-10-05:
**Description**

I have developed a message converter that can plot line and text. position of line and text are exactly the same but those are displayed in different positions

- Version: @foxglove/studio": "^1.53.1
- Platform: Desktop App
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

 

Position of text 
[image]

Position of line  
[image]

I believe that there is a bug regarding visualizing lines in 3D panel
**Expected Behavior**

jtbandes, 2023-10-05:
Could you please share sample data/code to reproduce the issue?

emresiimseek, 2023-10-06:
sure, 

```javascript
{
    "deletions": [],
    "entities": [
        {
            "id": "test",
            "timestamp": {
                "sec": 3885240,
                "nsec": 251390000
            },
            "frame_id": "test",
            "lifetime": {
                "sec": 10,
                "nsec": 0
            },
            "frame_locked": false,
            "metadata": [],
            "arrows": [
                {
                    "color": {
                        "r": 143,
                        "g": 188,
                        "b": 143,
                        "a": 255
                    },
                    "pose": {
                        "position": {
                            "x": -21.30497169494629,
                            "y": -0.20586971938610077,
                            "z": 0
                        },
                        "orientation": {
                            "x": 0,
                            "y": 0,
                            "z": 0,
                            "w": 1
                        }
                    },
                    "head_diameter": 0.5,
                    "head_length": 1,
                    "shaft_diameter": 0.3,
                    "shaft_length": 2
                }
            ],
            "cubes": [],
            "spheres": [],
            "cylinders": [],
            "lines": [
                {
                    "type": 0,
                    "color": {
                        "r": 143,
                        "g": 188,
                        "b": 143,
                        "a": 255
                    },
                    "colors": [],
                    "indices": [],
                    "points": [
                        {
                            "x": -21.898014068603516,
                            "y": 0.0862874984741211,
                            "z": 0
                        },
                        {
                            "x": -21.386249542236328,
                            "y": 0.4579019546508789,
                            "z": 0
                        },
                        {
                            "x": -21.00487518310547,
                            "y": 0.5785164833068848,
                            "z": 0
                        },
                        {
                            "x": -20.75387191772461,
                            "y": 0.4481353759765625,
                            "z": 0
                        },
                        {
                            "x": -20.633256912231445,
                            "y": 0.06675243377685547,
                            "z": 0
                        },
                        {
                            "x": -20.773405075073242,
                            "y": -0.8166260719299316,
                            "z": 0
                        },
                        {
                            "x": -20.964092254638672,
                            "y": -0.8769330978393555,
                            "z": 0
                        },
                        {
                            "x": -21.71708869934082,
                            "y": -0.48578548431396484,
                            "z": 0
                        },
                        {
                            "x": -21.898014068603516,
                            "y": 0.0862874984741211,
                            "z": 0
                        }
                    ],
                    "pose": {
                        "position": {
                            "x": -21.30497169494629,
                            "y": -0.20586971938610077,
                            "z": 0
                        },
                        "orientation": {
                            "x": 0,
                            "y": 0,
                            "z": 0,
                            "w": 1
                        }
                    },
                    "thickness": 0.1,
                    "scale_invariant": false
                }
            ],
            "triangles": [],
            "texts": [
                {
                    "color": {
                        "r": 143,
                        "g": 188,
                        "b": 143,
                        "a": 255
                    },
                    "pose": {
                        "position": {
                            "x": -21.30497169494629,
                            "y": -0.20586971938610077,
                            "z": 0
                        },
                        "orientation": {
                            "x": 0,
                            "y": 0,
                            "z": 0,
                            "w": 1
                        }
                    },
                    "scale_invariant": false,
                    "text": "ID:24",
                    "billboard": true,
                    "font_size": 1
                }
            ],
            "models": []
        }
    ]
}
```

emresiimseek, 2023-10-09:
Okay, I found out the problem If specifying the pose field the points become relative. When pose field changed as {x:0,y:0,z:0} points displayed exactly what I expected. However, in the documentation, it is not specified. As a feedback It can  specified clearly in the doc.

https://foxglove.dev/docs/studio/messages/line-primitive

closed 2023-10-09 by emresiimseek