foxglove/studio #6957: Plot panel loops back to the beginning of the plot

IamPhytan, 2023-10-05:
**Description**

When plotting time series in a plot panel, the time series seems to loop to the beginning of the plot

[image]

- Version: 1.72.0
- Platform: Ubuntu 22.04.3 LTS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS 1 bag

**Steps To Reproduce**

- Open a bag file as a local file
- Add a series in a plot panel and set its message path
- In the same panel, add another series and set the message path
- The panel shows a plot of the series that loops back to the beginning of the bag

**Expected Behavior**

The first and last point of the time series should not be connected unless the time series contains less than 3 points.

jtbandes, 2023-10-05:
Thanks, we are aware of this issue and are actively working on it.

cfoust, 2023-10-24:
Appears to be fixed!

closed 2023-10-24 by cfoust