foxglove/studio #6994: can't get TextAnnotation to show text on Foxglove Image pane

cdahms123, 2023-10-13:
**Description**
- Version: Foxglove 1.53.0 and 1.73.0
- Platform: Ubuntu 22.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): webcam video

**Steps To Reproduce**
I'm trying to get text to appear on a Foxglove Image pane, and ideally also on a 3D pane, but I'll start with the Image pane in this post.

I've referenced these documents:
https://foxglove.dev/docs/studio/panels/image
https://foxglove.dev/blog/annotate-your-robots-camera-images-with-image-markers
https://foxglove.dev/blog/introducing-text-annotations-in-foxglove-studios-image-panel
https://foxglove.dev/docs/studio/messages/text-annotation

I adapted https://foxglove.dev/blog/annotate-your-robots-camera-images-with-image-markers to ROS 2 Humble, then attempted to add text annodations as well as follows:

```
# my_webcam_publisher.py

import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image
from image_geometry.cameramodels import PinholeCameraModel
from cv_bridge import CvBridge
import cv2
import numpy as np

class ImagePublisher(Node):
    def __init__(self):
        super().__init__('my_webcam_publisher')
        self.publisher = self.create_publisher(Image, '/webcam/image', 10)
        self.camera_model = PinholeCameraModel()
        self.cv_bridge = CvBridge()

    def publish_image(self, image):
        height, width = image.shape[:2]
        print('width = ' + str(width) + ', height = ' + str(height))

        img_msg = self.cv_bridge.cv2_to_imgmsg(image, "bgr8")
        self.publisher.publish(img_msg)
        self.get_logger().info('Image published')

def main(args=None):
    rclpy.init(args=args)
    image_publisher = ImagePublisher()

    # if you only have 1 webcam make this parameter 0, otherwise you will need to experiment to find your webcam index
    cap = cv2.VideoCapture(0)
    if not cap.isOpened():
        print('could not open webcam')
        quit()
    # end if

    try:
        while rclpy.ok():
            ret, image = cap.read()
            if not ret:
                print('could not read frame from webcam')
                quit()
            # end if

            image_publisher.publish_image(image)
    except KeyboardInterrupt:
        pass

    image_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

```
# my_processor.py

import rclpy
from rclpy.node import Node
from cv_bridge import CvBridge
from sensor_msgs.msg import Image
from std_msgs.msg import ColorRGBA
import cv2

from foxglove_msgs.msg import ImageMarkerArray, TextAnnotation, TextMarker, TextPrimitive, Point2, Color
from visualization_msgs.msg import ImageMarker
from geometry_msgs.msg import Point

class ImageSubscriber(Node):
    def __init__(self):
        super().__init__('my_processor')
        self.subscription = self.create_subscription(Image, '/webcam/image', self.image_callback, 10)
        self.bridge = CvBridge()

        # publishers
        self.pub_markers = self.create_publisher(ImageMarkerArray, "/usb_cam/face_markers", 10)
        self.text_ann_publisher = self.create_publisher(TextAnnotation, '/webcam/my_text_ann', 10)
        self.text_marker_publisher = self.create_publisher(TextMarker, '/webcam/my_text_marker', 10)
        self.text_primitive_publisher = self.create_publisher(TextPrimitive, '/webcam/my_text_primitive', 10)
    # end function

    def image_callback(self, msg):
        try:
            image = self.bridge.imgmsg_to_cv2(msg, 'bgr8')

            height, width = image.shape[:2]
            print('received image, width = ' + str(width) + ', height = ' + str(height))

            # un-comment these 2 lines to show the image in a separate OpenCV window if desired
            # cv2.imshow("Received Image", cv_image)
            # cv2.waitKey(1)

            while_color_rgba = ColorRGBA()
            while_color_rgba.r = 1.0
            while_color_rgba.g = 1.0
            while_color_rgba.b = 1.0
            while_color_rgba.a = 1.0

            white_color = Color()
            white_color.r = 1.0
            white_color.g = 1.0
            white_color.b = 1.0
            white_color.a = 1.0

            black_color = Color()
            black_color.r = 0.0
            black_color.g = 0.0
            black_color.b = 0.0
            black_color.a = 1.0

            ### ImageMarker / ImageMarkerArray ####################################################

            point_top_left = Point()
            point_top_left.x = 200.0
            point_top_left.y = 200.0

            point_top_right = Point()
            point_top_right.x = 300.0
            point_top_right.y = 200.0

            point_bottom_right = Point()
            point_bottom_right.x = 300.0
            point_bottom_right.y = 300.0

            point_bottom_left = Point()
            point_bottom_left.x = 200.0
            point_bottom_left.y = 300.0

            image_marker = ImageMarker()
            image_marker.header = msg.header
            image_marker.scale = 1.0
            image_marker.type = ImageMarker.POLYGON
            image_marker.outline_color = while_color_rgba
            image_marker.points = [point_top_left, point_top_right, point_bottom_right, point_bottom_left]

            image_marker.fill_color = while_color_rgba
            image_marker.filled = True

            markers = ImageMarkerArray()
            markers.markers.append(image_marker)

            self.pub_markers.publish(markers)

            ### TextAnnotation ####################################################################

            text_pos = Point2()
            text_pos.x = 100.0
            text_pos.y = 100.0

            text_ann = TextAnnotation()
            text_ann.text = "abc 123 xyz"

            text_ann.background_color = black_color
            text_ann.text_color = white_color

            text_ann.font_size = 20.0
            text_ann.position = text_pos

            self.text_ann_publisher.publish(text_ann)

            ### TextMarker ########################################################################

            text_marker = TextMarker()
            text_marker.text = "abc 123 xyz"

            text_marker.color = white_color

            self.text_marker_publisher.publish(text_marker)

            ### TextPrimitive #####################################################################

            text_prim = TextPrimitive()

            text_prim.color = white_color
            text_prim.text = "abc 123 xyz"
            text_prim.font_size = 20.0
            
            self.text_primitive_publisher.publish(text_prim)

        except Exception as e:
            self.get_logger().error(f"Error processing image: {str(e)}")
        # end try
    # end function

# end class

def main(args=None):
    rclpy.init(args=args)
    image_subscriber = ImageSubscriber()
    rclpy.spin(image_subscriber)
    image_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

The white square drawn from 200, 200 to 300, 300 appears as expected, but none of the text annotations appear.  If I check for messages from the terminal with `ros2 topic echo /webcam/my_text_ann` I get:
```
---
timestamp:
  sec: 0
  nanosec: 0
position:
  x: 100.0
  y: 100.0
text: abc 123 xyz
font_size: 20.0
text_color:
  r: 1.0
  g: 1.0
  b: 1.0
  a: 1.0
background_color:
  r: 0.0
  g: 0.0
  b: 0.0
  a: 1.0
---
timestamp:
  sec: 0
  nanosec: 0
position:
  x: 100.0
  y: 100.0
text: abc 123 xyz
font_size: 20.0
text_color:
  r: 1.0
  g: 1.0
  b: 1.0
  a: 1.0
background_color:
  r: 0.0
  g: 0.0
  b: 0.0
  a: 1.0
---
.
.
.
```
So I know the text annotation message is being sent but I can't get it to appear in Foxglove.  I've tried this with both Foxglove 1.53.0 and 1.73.0, see attached screenshots of the Image Panel Settings for both versions.  As you can see the `/usb_cam/face_markers` topic option appears (this is for `ImageMarker` / `ImageMarkerArray`) but the `/webcam/my_text_ann` topic (for `TextAnnotation`) does not appear.  Is this a bug or am I missing a setting somewhere?

**Expected Behavior**
Getting a text annotation to appear in a Foxglove Image pane<br>
[image]
[image]

achim-k, 2023-10-24:
@cdahms123 could you provide a mcap or ROS bag file that includes image + text annotation messages that we can use to reproduce this issue?

> Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): webcam video

`webcam video` is not a valid data source. Do you use foxglove bridge or another live connection (rosbridge, ROS 1)?

achim-k, 2023-10-30:
> The white square drawn from 200, 200 to 300, 300 appears as expected, but none of the text annotations appear. If I check for messages from the terminal with `ros2 topic echo /webcam/my_text_ann` I get:
> 
> ```
> ---
> timestamp:
>   sec: 0
>   nanosec: 0
> position:
> ```

@cdahms123 the timestamps of the text annotation messages are zero. Quite sure that this is your issue. They should ideally match the timestamps of the corresponding images.

achim-k, 2023-11-03:
Closing this issue due to inactivity. @cdahms123 feel free to open again if you are still affected by this.

closed 2023-11-03 by achim-k