foxglove/studio #6998: 3D Panel URDF not Visualized Properly

peterchenyipu, 2023-10-17:
**Description**
Loading URDF visual meshes or collisions are different from the results in RVIZ2.

- Version: 1.73.0
- Platform: Ubuntu 22
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ros-foxglove-bridge

**Steps To Reproduce**
I have tried many URDFs across different robot's descriptions projects. Here is an example:
1. Clone and build https://github.com/aica-technology/kuka-iiwa-description/tree/main
2. Launch [visualize.launch.py](https://github.com/aica-technology/kuka-iiwa-description/blob/main/launch/visualize.launch.py)
3. Use foxglove studio to open a websocket connection, add a 3D panel and turn on the robot_description view under Topics.

**Expected Behavior**
A correct robot visualization should be shown. However on my system the meshes are not aligned properly for both visual and collision.
[image]
[image]

achim-k, 2023-10-18:
Can you try setting the [Mesh up axis](https://foxglove.dev/docs/studio/panels/3d#scene) setting to `Z-up`? The setting can be found in the 3D panel settings under `scene`.

Studio uses `Y-up` by default which is different from what Rviz uses (`Z-up`)

peterchenyipu, 2023-10-18:
> Can you try setting the [Mesh up axis](https://foxglove.dev/docs/studio/panels/3d#scene) setting to `Z-up`? The setting can be found in the 3D panel settings under `scene`.
> 
> Studio uses `Y-up` by default which is different from what Rviz uses (`Z-up`)

That worked! Thank you so much!

closed 2023-10-18 by peterchenyipu