foxglove/studio #7007: Flickering bugs for GLTF models

henrik-kaijser, 2023-10-20:
**Description**
I have implemented a message converter that produces a height map as a GLTF in a ModelPrimitive message, which is sent in a SceneUpdate Message. When the model is displayed in the studio 3D panel there's substantial flickering. 

- Version: latest studio
- Platform: ubuntu 22.04
- Data source: mcap

**Studio Code Analysis**
The `#updateModels` method in `ThreeDeeRender/renderables/primitives/RenderableModels.ts` begins by doing
`this.clear() `
which wipes the currently loaded model from Three. Then it starts loading the next version of the model  asynchronously. Since this may take quite some time, a new frame that is rendered before the model has been loaded will this not contain the model, which causes flickering.
Similarly, after the model has been successfully loaded towards the end of  `#updateModels` it gets rid of the model by doing
```
renderable.model.removeFromParent();
this.#disposeModel(renderable);
```
which creates a very similar problem.

**Workaround**  
As a quick hack I removed the three lines listed above, and instead cleared the previous model version after the new version has been loaded, but before
_this.add(renderable.model);_ 
This fixes the problem but you may want to fix it in some other way.

defunctzombie, 2023-10-24:
Internal tracking ticket: https://linear.app/foxglove/issue/FG-5384/flickering-bugs-for-gltf-models

snosenzo, 2023-11-02:
The lines that are disposing of the renderable in the `.finally` should be kept to dispose of unused previous renderables. The renderables that are used across updates should be taken out of that `prevRenderables` list by the time it gets to the finally. I moved some code around and renamed some things to hopefully make that a bit more clear. 

I tested with some sample data of my own and didn't have a flicker. Let me know if this fix works for your case.

closed 2023-11-08 by snosenzo