foxglove/studio #7021: Tab subpanel is cleared when switching tabs during live connection

ilmagico, 2023-10-24:
**Description**

During a live connection, when a panel that accumulates data (e.g. Plot or Log) is used inside a Tab panel, all the data accumulated is cleared upon switching tabs. For example, a Plot is cleared and the plot line restarts only with newly received data, or a Log is cleared and only new messages are displayed.

- Version: 1.74.1
- Platform: Linux
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS2 connection via foxglove bridge

**Steps To Reproduce**

1. Connect to a live data source
2. Create a Tab panel
3. Create at least two tabs, say, a Plot and a Log tabs (pick some interesting topics to show)
4. Observe the plot (or log), then switch tab and switch back

**Expected Behavior**

The same data is show in the plot tab after switching back  (or the same text is shown in the log).

** Actual Behavior **

The Plot (or Log) tab is completely cleared after switching back to it.

ilmagico, 2023-10-24:
https://github.com/foxglove/studio/assets/7019416/963dd6a8-53a6-4ae4-bbff-c253536d3e42

jtbandes, 2023-10-25:
This is not a bug, because the app is behaving as designed. That said, it is evidently a pain point for some users and we will consider possible solutions. Closing as duplicate of https://github.com/orgs/foxglove/discussions/846

closed 2023-10-25 by jtbandes