foxglove/studio #7054: Using “main”: “./dist/entrypoint.js” fails at runtime

jhurliman, 2023-10-31:
**Description**

In package.json, using “main”: “./dist/entrypoint.js” instead of “dist/entrypoint.js” fails to load the extension (even though the entry point is valid and recognized by node) because the map of files returned by JSZip includes an entry for “dist/entrypoint.js” but not the dot-prefixed variant.

**Expected Behavior**

Use path normalization when searching for the entrypoint file in the zip instead of exact string matching.

defunctzombie, 2023-10-31:
Originally filed in the `create-foxglove-extension` repo. I moved this to the studio repo which handles the loading of the extension. We might learn that we need to fix this in the packaging but my initial hunch is we can solve it in studio.

defunctzombie, 2023-11-21:
Thanks for surfacing. The fix for this will be in the next release.

closed 2023-11-21 by defunctzombie