foxglove/studio #7127: Foxglove User Script show print NavSatFix does not work anymore

kis-balazs, 2023-11-16:
Trying to read GPS signal from a topic, and convert it into NavSatFix to plot on map panel. Does not work, it even says from time to time that topic - which clearly exists in the topic list - does not exist. 

- Version: 1.69.0
- Platform: Windows
- Data source ROS1 bag file:

import { Input, Messages } from "ros";

export const inputs = ["/my/gps/topic"];
export const output = "/studio_node/gps";

const pointPublisher = (
  message: Input<"/my/gps/topic">
): Messages.sensor_msgs__NavSatFix => {
  return {
    header: message.message.header,
    status: { status: 1, service: 1 },
    latitude: message.message.Lat,
    longitude: message.message.Lon,
    altitude: 0,
    position_covariance: [],
    position_covariance_type: 0,
  };
};

export default pointPublisher;

Create topic of name specified in the variable output and plot the respective GPS coordinates for every time a GPS coordinate in the input topic is received.

defunctzombie, 2023-11-16:
Could you also provide a sample file that you are using with the above script?

kis-balazs, 2023-11-17:
Unfortunately I cannot, but it is a GPS sensor sending data at 25Hz, and we have a recording of this, using a custom ROS message.

defunctzombie, 2023-11-18:
I need a sample file that is causing the problem to reproduce this.

Another thing you could try is to update to the latest version of the app. 1.69 is several versions behind. If that does not work please provide a sample file.

achim-k, 2023-11-21:
@kis-balazs I quickly tried this out by writing a user script that converts a `geometry_msgs/TransformStamped` to a `sensor_msgs/NavSatFix` message. It works as expected, see the screencast below. 

Some remarks:
- Note the `new Float64Array()` for `position_covariance` (user script panel raises an error on empty array)
- Make sure you have saved the UserScript
- Make sure that your input topic is published by inspecting it with e.g. the RawMessage panel

[Screencast from 21.11.2023 10:12:20.webm](https://github.com/foxglove/studio/assets/9250155/dc1a9e70-d0e6-4295-b2bd-ca578098a8c9)

Does this solve your problem?

closed 2023-11-21 by achim-k