foxglove/studio #7129: Worker crashes when loading custom font in Safari

zeeyang, 2023-11-16:
**Description**
Plot and state transition panels are blank in iOS Safari. This is caused by worker crash when loading custom font (CORS error when loading data blob in worker). This looks like a Safari bug but Foxglove should also handle the error without crashing worker.

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/components/Chart/worker/ChartJsMux.ts#L61-L69

- Version: 1.77.0 and prior versions
- Platform: iOS and macOS Safari 17
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap

**Steps To Reproduce**

- load Foxglove web in Safari 17
- create plot panel or state transition panel
- verify raw data panel is updating but plot panels are blank

jtbandes, 2023-11-17:
I can reproduce the problem with this code:

```ts
new Worker(URL.createObjectURL(new Blob([`(async()=>{
  try {
    await new FontFace("ABC", "url(data:font/woff2,abcd) format('woff2')").load().catch(e=>console.error("rejected",e));
    console.log("done")
  } catch(e) {
    console.error("caught",e)
  }
})()`])))
```

[image]

However I'm not sure if there is any way to catch the error since my attempts above don't seem to work (I don't see any further logs after the failure). If you can find any fix or workaround, let me know, but since we do not claim to support Safari, we may end up closing this ticket. If you think this is a true bug it could be worth filing the bug with Safari.

jtbandes, 2023-11-17:
This seems to work: https://github.com/foxglove/studio/pull/7139

jtbandes, 2023-11-17:
Filed https://bugs.webkit.org/show_bug.cgi?id=265000

zeeyang, 2023-11-17:
Thanks for looking into this. I just tested it - yes you are right, try catch was not sufficient to stop the worker from crashing. I would never think to use bytearray with fontface. Thank you for the fix!

closed 2023-11-17 by zeeyang