foxglove/studio #7148: Pointcloud2 visualization

edvart-ros, 2023-11-20:
- Version: Recently installed foxglove studio, ROS2 Iron
- Platform: Docker on windows
- Data source: Live connection to container

Attempted to visualize the pointcloud2 message in foxglove studio. Foxglove complains about "PointCloud must contain at least two of x/y/z fields" and wont visualize the points. I am manually populating these messages, but have verified the format by visualizing the messages in rviz2. Rviz has no issues with the message format and does the visualization correctly. Is this just a foxglove bug or is it being intentionally more picky about the format?

Example message:
```
{
  "header": {
    "stamp": {
      "sec": 1700177019,
      "nsec": 615725994
    },
    "frame_id": "wamv/lidar_link"
  },
  "height": 1,
  "width": 1728,
  "fields": [
    {
      "name": "x",
      "offset": 0,
      "datatype": 7,
      "count": 1728
    },
    {
      "name": "y",
      "offset": 4,
      "datatype": 7,
      "count": 1728
    },
    {
      "name": "z",
      "offset": 8,
      "datatype": 7,
      "count": 1728
    }
  ],
  "is_bigendian": false,
  "point_step": 12,
  "row_step": 20736,
  "data": [
    0,
    0,
    192,
    255,
    0,
    0,
    192,
    127,
    ...
  ],
  "is_dense": true
}
```
[image]

[image]

jtbandes, 2023-11-20:
Please share a bag/mcap file that we can use to reproduce the issue.

edvart-ros, 2023-11-20:
[rosbag2_2023_11_20-18_55_07_0.zip](https://github.com/foxglove/studio/files/13418104/rosbag2_2023_11_20-18_55_07_0.zip)

jtbandes, 2023-11-20:
The problem is the `"count": 1728` – foxglove only supports fields with count equal to 1.

closed 2023-11-20 by jtbandes