foxglove/studio #7182: Foxglove Studio is not starting on a Jetson Xavier NX

aurelien-m, 2023-11-28:
**Description**

I just installed Foxglove Studio from the [download page](https://foxglove.dev/download) and the installation went perfectly fine. But when I try to launch the software, it does absolutely nothing.

- Version: 1.79.0
- Platform: Ubuntu 20.04 64-bit ARM
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS 1 native

**Steps To Reproduce**

I tried launching it from the terminal:
```
$ foxglove-studio
# nothing is happening here
```

I also tried `--help` and `--version` but it does nothing. I also tried launching from the application menu but the same issue happens.

achim-k, 2023-12-01:
Can you run any other Electron app without crashing? I believe Visual Studio Code is also an Electron app. Alternatively you can test if you can run a electron fiddle: https://www.electronjs.org/fiddle

achim-k, 2023-12-12:
@aurelien-m did you solve your problem?

aurelien-m, 2023-12-12:
Hi @achim-k, unfortunately no, VS Code does indeed need to be tweaked to work on the Jetson (see https://github.com/microsoft/vscode/issues/162525). I'm not sure if it's an electron related issue, but I've been able to work around it by using the foxglove bridge and opening Foxglove on an other computer (which is fine and we can close the issue if you want)

achim-k, 2023-12-12:
Ok, I'm going to close this since this seems to be an Electron issue. Note that you can always open https://studio.foxglove.dev in a browser.

closed 2023-12-12 by achim-k