foxglove/studio #7220: Image Visualization

peterdavidfagan, 2023-12-08:
**Description**
I am adding this issue for quick feedback in the event that it is a known error. If not I can pass further details of our setup.

Foxglove studio appears to display image differently to rviz2. 

[image]

**Steps To Reproduce**
Requires Zed2i camera with default camera config.  

**Expected Behavior**
Image display matches between rviz2 and foxglove.

defunctzombie, 2023-12-08:
@peterdavidfagan Please attach an mcap file with some of the topic image data to this issue so we can reproduce. It is hard to say more without some sample data to identify if its a data issue or a rendering issue.

jtbandes, 2023-12-12:
Note that (if I remember correctly) rviz's image display does not support camera distortion parameters.

defunctzombie, 2024-01-03:
Closing this for inactivity. If you are able to provide some more reproduction steps and sample file we can re-open and look into it.

closed 2024-01-03 by defunctzombie

peterdavidfagan, 2024-01-03:
Thanks @defunctzombie and apologies for inactivity, I haven't touched this item since we I am now working on simulations of experiments. I will get back to this in future when running real robot experiments.