foxglove/studio #7232: Windows 11 puts Foxglove into "efficiency mode"

andraspalffy, 2023-12-13:
**Description**
Windows 11 puts Foxglove into efficiency mode. This significantly throttles the performance. Even if I turn it off by Task Manager -> right click on Foxglove -> untick efficiency mode, it turns it back on after a couple of seconds.

[image]

What is worse that the issue seems to often be relevant for browsers, but then the blame is pushed to the program, allegedly "asking" to be throttled:

[https://www.reddit.com/r/Windows11/comments/133n9kw/can_i_prevent_windows_from_putting_certain_edge/
](https://www.reddit.com/r/Windows11/comments/133n9kw/can_i_prevent_windows_from_putting_certain_edge/
)
- Version: 1.81.0
- Platform: Windows 11
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbag

**Steps To Reproduce**
Take a large (100 GB) rosbag, and load it, play it with half speed or so.

**Expected Behavior**
I want Foxglove to run at full power, only limited by the capabilities of my hardware, not by the random, unasked for requests of Windows.

jtbandes, 2023-12-14:
Internal tracking issue: FG-6107

defunctzombie, 2024-02-26:
Thanks for reporting. We've added a flag that should prevent this. It will be in the next desktop release.

closed 2024-02-26 by defunctzombie