foxglove/studio #7260: Foxglove Studio takes too long to start

swarajpeppermint, 2023-12-19:
**Description**
Since a few releases ago, Foxglove Studio takes too long to start up (white screen to home screen). Attaching a screen recording of the delay. During the screen recording, I just have the browser running on the system.

- Version: 1.82.0
- Platform: Ubuntu 22.04.3
- System Specs: RAM 16.0 GB, Processor 11th Gen Intel® Core™ i7-11800H @ 2.30GHz × 16
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): N/A

**Steps To Reproduce**
Start Foxglove Studio on an updated Ubuntu 22.04 system

**Screen Recording**
[Screencast from 12-19-2023 09:43:18 AM.webm](https://github.com/foxglove/studio/assets/85288063/8c04d904-3bac-47a7-b386-035a4e3cf69a)

**Expected Behavior**
Should show home screen within a second or two

jtbandes, 2023-12-20:
Can you press ctrl+shift+i when the window appears and see if there is any information in the developer console or network tab?

swarajpeppermint, 2023-12-21:
[image]

swarajpeppermint, 2023-12-21:
Foxglove Studio started hanging after a while
[image]

swarajpeppermint, 2023-12-21:
network tab
[image]

playerdefault, 2024-02-22:
@wimagguc Yes I have a custom extension installed. It's this extension: https://github.com/playerdefault/foxglove-pointstamped-viz

playerdefault, 2024-02-22:
A good update on this. Launching foxglove studio via `foxglove-studio --disable-gpu --disable-software-rasterizer` launches it **instantly**. The same issue is happening with VSCode too. 
Relevant GitHub issue with VSCode: https://github.com/microsoft/vscode/issues/192493#issuecomment-1937098984

Maybe it's not an issue with FoxGlove but something to do with a recent update with Electron?

defunctzombie, 2024-02-26:
Folks experiencing this issue - did you install via Snap or Deb?

playerdefault, 2024-02-27:
> Folks experiencing this issue - did you install via Snap or Deb?

snap

defunctzombie, 2024-02-27:
@playerdefault if you are still experiencing this with the snap install - could you give the Deb install a try and see if it exhibits the same behavior on your machine. That would be a helpful data point for us.

playerdefault, 2024-03-01:
> @playerdefault if you are still experiencing this with the snap install - could you give the Deb install a try and see if it exhibits the same behavior on your machine. That would be a helpful data point for us.

issue is not occurring with the deb version