foxglove/studio #7278: The plot component has a problem displaying the chart when adding a curve.

Woodii1998, 2023-12-28:
**Description**
Plot component curves are lost when adding curves
The plot component loses some curves when adding curves.

- Version: online version
- Platform: web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap

**Steps To Reproduce**

https://github.com/foxglove/studio/assets/33593572/cac96841-799c-4d4c-a160-ba1068b9c7f5

**Expected Behavior**
When you add a curve, you can show all the curves.

Woodii1998, 2023-12-28:
mcap file is here, if you want

https://drive.google.com/file/d/1K042_0shfzOBv4FkieIsbqL-LMIIJ-mp/view?usp=sharing

defunctzombie, 2024-01-11:
@Woodii1998 could you give this preview deployment a try: https://studio-hdnbgt93s.foxglove.party/ I think it will fix your issue

Woodii1998, 2024-01-11:
> @Woodii1998 could you give this preview deployment a try: https://studio-hdnbgt93s.foxglove.party/ I think it will fix your issue

Looking good, looks like the issue is fixed!

closed 2024-01-13 by defunctzombie