foxglove/studio #7315: Invalid typed array lengt when opening mcap file

acschroeder, 2024-01-10:
Our System creates snapshots, so we can analyse problems later. But MarkerArrays and some other Messages like logs can not be read by Foxglove. The provided mcap file can be read by mcap cli but not by foxglove. How can we find the problem?

```
$> mcap info test.mcap
library: mcap go v1.0.3; libmcap 0.8.0
profile: ros2
messages: 35
duration: 1.92s
start: 10.040000000
end: 11.960000000
compression:
	zstd: [1/1 chunks] [9.99 KiB/7.11 KiB (28.83%)] [3.70 KiB/sec] 
channels:
  	(8) /rosout  35 msgs (18.23 Hz)   : rcl_interfaces/msg/Log [ros2msg]  
attachments: 0
metadata: 0
```

- Version: **1.83.1**
- Platform: **docker/web** and **ubuntu/snap**
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): **mcap**

**Steps To Reproduce**
* load provided mcap file in foxglove https://file.io/Fb7jT4IB4wDn
* ensure log panel is visible in foxglove
* start playback, error should occure in the problems section
[image]

**Expected Behavior**
* mcap file can be loaded and played back without any errors
* messages can be displayed and data can be used in the different panels

jtbandes, 2024-01-10:
How is your ROS system set up and how did you create this mcap file?

jmachowinski, 2024-01-11:
Bagfile was created using this application https://github.com/cellumation/snapshotter/tree/ros2

Our ROS is an ros2 iron.

jtbandes, 2024-01-11:
The error message makes it sound like the message data has been corrupted by the snapshotter. If you use `ros2 bag play` do you see the same errors? What if you try saving the snapshot into a db3 bag instead of mcap?

jmachowinski, 2024-01-12:
ros2 bag doctor and ros2 bag play do not show any errors / outputs. rqt_bag also opens the bag without problems. Therefore we don't think the file is corrupted. 

Might it be, that the bag does not contain some metadata, that is crucial for studio ?
Or that the typenames for the topics are off ?

jmachowinski, 2024-01-12:
@acschroeder The mcap file is not available any more on flie.io, please repost it

arneboe, 2024-01-16:
This mcap file uses the ros 2 message compression. Could this be a problem?
I.e. the corresponding ros2 metadata.yaml contains:
`
  compression_format: zstd
  compression_mode: MESSAGE
`

arneboe, 2024-01-16:
We changed the implementation of the snapshotter to use mcap's internal compression instead of ros message compression. This "fixes" this bug.

But it might still be a good idea to support ros compressed messages in foxglove (at least if a metadata.yaml is present).

jtbandes, 2024-01-16:
👍 Indeed, mcap is not compatible with `--compression-mode message`. Looks like there is a relevant error message that should be encountered, but maybe not in all circumstances: https://github.com/ros2/rosbag2/blob/317286c98472205593d5b0a791d06fea42613ba3/rosbag2_storage_mcap/src/mcap_storage.cpp#L845-L849  Foxglove does not support loading metadata.yaml files and so has no way to know whether message compression was used.

closed 2024-01-16 by jtbandes