foxglove/studio #7414: Plots with set time window x-axis range don't automatically scroll

davidzimmerjr, 2024-01-31:
**Description**

- Version: 1.86.0
- Platform: Mac OS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file

**Steps To Reproduce**

When plotting data and setting up the plot window with the below configuration where X axis has a set Range (seconds) [screenshot shows 15], in 1.84.0 the plot would scroll as the bag file would play.
[image]

**Expected Behavior**
Plot scrolls as the bag file plays.

defunctzombie, 2024-01-31:
Thanks for reporting this. It is fixed in 1.86.1

closed 2024-01-31 by defunctzombie

davidzimmerjr, 2024-01-31:
Yes! Thank you :) Just noticed it working!