foxglove/studio #7416: Compressed Image Display not working with latest nvidia-drivers

joshi-bharat, 2024-01-31:
**Description**
The image display is not working after upgrading nvidia drivers to 535

- Version: 1.86.1
- Platform: Ubuntu 22.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): custom websocket based streaming

**Steps To Reproduce**

The image display is not working after updating to nvidia-driver-535. Works fine with driver-525. 
[image]
As you can see the foxglove is getting the images and I can download the image correctly. The only problem is with the display.

**Expected Behavior**
I am getting following error in terminal launching foxglove-studio sporadically.
`Warning: loader_get_json: Failed to open JSON file lvp_icd.x86_64.json
Warning: /usr/lib/x86_64-linux-gnu/libvulkan_intel.so: cannot open shared object file: Permission denied
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/x86_64-linux-gnu/libvulkan_intel.so. Ignoring this JSON
Warning: loader_get_json: Failed to open JSON file intel_hasvk_icd.x86_64.json
Warning: /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so: cannot open shared object file: Permission denied
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so. Ignoring this JSON
Warning: loader_scanned_icd_add: Could not get 'vkCreateInstance' via 'vk_icdGetInstanceProcAddr' for ICD libGLX_nvidia.so.0
Warning: loader_get_json: Failed to open JSON file nvidia_layers.json
Warning: vkCreateInstance: Found no drivers!
Warning: vkCreateInstance failed with VK_ERROR_INCOMPATIBLE_DRIVER
    at CheckVkSuccessImpl (../../third_party/dawn/src/dawn/native/vulkan/VulkanError.cpp:101)
    at CreateVkInstance (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:493)
    at Initialize (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:379)
    at Create (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:301)
    at operator() (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:556)
`

jtbandes, 2024-01-31:
How did you install Foxglove – from the snap or deb? If it was the snap, please try installing from the deb directly.

joshi-bharat, 2024-01-31:
I installed using  debian.

jtbandes, 2024-01-31:
The following issues seem related:
- https://github.com/ValveSoftware/steam-for-linux/issues/9630
- https://forums.developer.nvidia.com/t/no-hardware-acceleration-in-google-chrome-with-535-x-drivers/269398
- https://bugs.launchpad.net/ubuntu/+source/nvidia-graphics-drivers-535/+bug/2026295

What do you see when visiting `about:gpu` in chrome?

What do you see when opening the developer tools in foxglove (ctrl+shift+i)? Any relevant errors in there?

joshi-bharat, 2024-02-01:
I have attached the output of about:gpu in the text file
[about-gpu.txt](https://github.com/foxglove/studio/files/14129461/about-gpu.txt)

Also, floxglove is complaining about not finding sharers. See the attached log below.
`main.js:5862 WebGL: INVALID_OPERATION: drawElements: no valid shader program in use
v1 @ main.js:5862
V6.renderBufferDirect @ main.js:5920
J9 @ main.js:5920
v9 @ main.js:5920
Nm @ main.js:5920
V6.render @ main.js:5920
#k @ 2890.js:345
animationFrame @ 2890.js:345
F @ main.js:6323
requestAnimationFrame (async)
(anonymous) @ main.js:6326
queueAnimationFrame @ 2890.js:345
(anonymous) @ main.js:2051
Promise.then (async)
setImage @ main.js:2051
setImage @ main.js:6423
#d @ main.js:2051
#p @ main.js:2051
(anonymous) @ main.js:2051
#c @ main.js:2051
handleImage @ main.js:2051
handleCompressedImage @ main.js:2051
#z @ 2890.js:345
#k @ 2890.js:345
animationFrame @ 2890.js:345
(anonymous) @ 2890.js:345
f4 @ main.js:1412
hc @ main.js:1412
Po @ main.js:1412
c3 @ main.js:1410
Ei @ main.js:1412
c4 @ main.js:1410
(anonymous) @ main.js:1410
(anonymous) @ main.js:6311
d @ main.js:6311
dispatch @ main.js:1630
listener @ main.js:1630
#R @ main.js:6407
await in #R (async)
processPlayerState @ main.js:6407
(anonymous) @ main.js:1605
(anonymous) @ main.js:1507
(anonymous) @ main.js:2168
(anonymous) @ main.js:2168
L @ main.js:2168
Promise.then (async)
T @ main.js:2168
(anonymous) @ main.js:2168
v @ main.js:2168
runExclusive @ main.js:2168
runExclusive @ main.js:2168
runExclusive @ main.js:1507
#s @ main.js:1605
(anonymous) @ main.js:1605
(anonymous) @ main.js:1633
g @ main.js:1507
(anonymous) @ main.js:1507
Promise.finally (async)
g @ main.js:1507
b @ main.js:1507
(anonymous) @ main.js:1633
f.emit @ main.js:1320
#t.onmessage @ main.js:1632
Ye.#e.onmessage @ main.js:1632
main.js:5888 THREE.WebGLProgram: Shader Error 0 - VALIDATE_STATUS false

Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Dec 28 2023. Please check build information of source that generated the binary.
Location of variable pc_fragColor conflicts with another variable.
A4 @ main.js:5888
`

jtbandes, 2024-02-01:
Also sounds like this error. https://github.com/foxglove/studio/issues/6706 (https://github.com/electron/electron/pull/40450)

jtbandes, 2024-02-01:
The title of https://github.com/electron/electron/pull/40450 implies it is a problem with a cached shader, so you could try deleting app data and caches, reinstalling the app, etc. You could also try using Chrome and see if it helps. But there is unfortunately probably not much we can do on the Foxglove side to help resolve it if the root cause is upstream in Chrome/Electron.

joshi-bharat, 2024-02-01:
Okay.

jtbandes, 2024-02-12:
I'm going to close out this ticket because I'm not aware of anything we can do here other than trying the steps recommended above. If it continues to be a problem feel free to continue posting updates here and we can try to help.

closed 2024-02-12 by jtbandes