foxglove/studio #7459: Visualizing ROS1 mcap files is sluggish

t-yep-cpr, 2024-02-07:
**Description**
Visualizing ROS1 mcap files is sluggish. I've tried uploading an mcap file that was converted from a ROS1 bag into our(otto) test instance (studio.foxglove.dev) with foxglove. The playback fps is very low and I'm unable to play the bag or do anything useful with this file on this instance. I've attached a video for this. Would love to get any feedback on what's going on here. 

- Version: 1.86.1+76a7d40.128b9d5
- Platform: Foxglove-Studio
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):ROS1 converted mcap file. (TY_AMR_4_4AED6E44_base.webviz.mcap)
- Chrome browser version: Version 109.0.5414.119 (Official Build) (64-bit)
  - Use HW acceleration option enabled

**Steps To Reproduce**

Uploaded a ROS1 bag that was converted into mcap and tried to play it. 
[sluggish.webm](https://github.com/foxglove/studio/assets/159342923/36e034d6-3456-4023-b763-0ce1ec971994)

**Expected Behavior**
Smooth playback

defunctzombie, 2024-02-08:
@t-yep-cpr Thanks for sharing the details here. I tried to reproduce using the referenced file and layout but I am seeing smooth playback on my M1 Macbook Pro on Chrome: 121.0.6167.139 (Official Build) (arm64).

If you can upgrade chrome please try that and see if it helps at all? The version of Chrome you are on seems outdated (even in your screencap it wants to update).

If upgrading chrome does not work we can certainly look closer into why your playback performance is worse but would likely need to do some pairing if we cannot reproduce on our computers.

t-yep-cpr, 2024-02-14:
Thanks for responding so quickly, @defunctzombie.  
I'll be sure to link any issues similar to this in a future ticket should I come across any. 
I did notice that after clearing cache data in Chrome, it did improve playback slightly.

closed 2024-02-14 by t-yep-cpr