foxglove/studio #7497: Unable to visualize meshes from URDF in URDF plugin

srsidd, 2024-02-19:
**Description**
When running the FoxGlove studio desktop app, the meshes specified in the URDF of a UR robot do not get rendered, and throw an error instead - `Failed to load mesh from  "package://ur_description/meshes/ur5e/visual/base.dae"`.  This could be possibly related to #4808.

- Version: Foxglove Studio 1.87.0
- Platform: Ubuntu 22.04.3 LTS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): N/A

**Steps To Reproduce**
* Launch the UR robot with `ros2 launch ur_description view_ur.launch.py ur_type:=ur5e`. This brings up rviz, where the robot model is visualized - 

[image]

* Launch the ROS2 web bridge - ros2 launch rosbridge_server rosbridge_websocket_launch.xml
* Create a URDF file from the Xacro file - `xacro /opt/ros/humble/share/ur_description/urdf/ur.urdf.xacro name:=ur ur_type:=ur5e > /home/user/ur.urdf`
* Launch FoxGlove desktop app, connect to web bridge, drop down custom layers and select add urdf. Select the `Source: File Path (Desktop only)`  option, and under `File path: /home/sidd/Documents/ur.urdf` paste the path from the previous command where the URDF was created. Once done, you will be able to see all the tf's being visualized but the robot model does not. 
[image]

Hovering over the red exclamation warning sign next to URDF shows the message - `Failed to load mesh from  "package://ur_description/meshes/ur5e/visual/base.dae"`.

**Expected Behavior**
I'm expecting to see the robot model displayed in FoxGlove like I do in Rviz.

achim-k, 2024-02-21:
You will have to tell the Foxglove Desktop app where it can find mesh files referenced by your URDF. To do that, you have to specify the [`ROS_PACKAGE_PATH`](https://docs.foxglove.dev/docs/visualization/introduction/#general) in the app settings. A typical value would be something like `/path/to/workspace/src:/opt/ros/noetic/share`

closed 2024-02-21 by achim-k

srsidd, 2024-02-24:
I have the `ROS_PACKAGE_PATH` variable set and pointing to `/opt/ros/humble/share` - 
[image]

I see the same error. I tried loading the urdf from `/opt/ros/humble/share/ur_description/urdf` and `/home/sidd/Documents/` but they both result in the same error.

achim-k, 2024-02-26:
Have you restarted the app after setting the `ROS_PACKAGE_PATH`?
What data source are you using? Do you also see this error also if you don't connect to any data source at all?