foxglove/studio #7530: Unable to view live data in Chrome on Android

craigblaster, 2024-03-08:
**Description**
Connecting to a live ROS2 data source from my Android phone running Chrome gives me a prompt to enable "unsafe scripts" using the shield icon. Screenshot attached (in comment).
There is no shield icon or similar setting that I could find.

- Version: web
- Platform: Android / Chrome 122
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS2 native

**Steps To Reproduce**
1. Run the foxglove bridge on a ROS2 machine.
2. Using the webapp app.foxglove.dev, choose 'open connection' and enter the URL/IP
3. Error message is shown

**Expected Behavior**
* Show the correct instructions to enable the live data view

craigblaster, 2024-03-08:
[image]

craigblaster, 2024-03-08:
> Are you able to find relevant settings by following these instructions?
> https://support.google.com/chrome/answer/114662?hl=en&co=GENIE.Platform%3DAndroid

None of the settings or permissions there seem related to the problem.
I have set everything to Allowed or Ask but the problem persists.

chfritz, 2024-03-08:
One thing you could try is to open about:flags in your browser and add your ws:// url to the list of "Insecure origins treated as secure".